QuantumSavory/QuantumSavory.jl #41: Incorrect results when `H` is applied to 2nd qubit of a 2 qubit system

Abhishek-1Bhatt, 2023-09-10:
```julia
using QuantumSavory

ra = Register(1)
rb = Register(1)

st = L0⊗X1

initialize!((ra[1], rb[1]), st)
ra.staterefs[1].state[]

apply!(rb[1], H)
ra.staterefs[1].state[]
```
Here, the initial state `L0⊗X1` should be transformed to `L0⊗L0` when Hadamard is applied to the 2nd qubit, but instead we get `X1⊗X1`

Krastanov, 2023-09-11:
Could you double check this? I do not seem to be able to reproduce it:

```
julia> ra = Register(1)
Register with 1 slots: [ Qubit ]
  Slots:
    nothing

julia> rb = Register(1)
Register with 1 slots: [ Qubit ]
  Slots:
    nothing

julia> st = L0⊗X1
|Z₁⟩|X₁⟩

julia> initialize!((ra[1], rb[1]), st)
State containing 2 subsystems in QuantumOpticsBase implementation
  In registers:
    1@12899361520081501929
    1@4465442880340761800

julia> ra.staterefs[1].state[]
Ket(dim=4)
  basis: [Spin(1/2) ⊗ Spin(1/2)]
 0.7071067811865475 + 0.0im
                0.0 + 0.0im
 0.7071067811865475 + 0.0im
                0.0 + 0.0im

julia> apply!(rb[1], H)
(Register[Register with 1 slots: [ Qubit ]
  Slots:
    Subsystem 2 of QuantumOpticsBase.Ket 16783222433556155838], 0.0)

julia> ra.staterefs[1].state[]
Ket(dim=4)
  basis: [Spin(1/2) ⊗ Spin(1/2)]
 0.9999999999999999 + 0.0im
                0.0 + 0.0im
                0.0 + 0.0im
                0.0 + 0.0im
```

Please reopen if it is indeed reproducible.

closed 2023-09-11 by Krastanov

Abhishek-1Bhatt, 2023-09-12:
Oh, my bad, I was testing it on a wrong branch