QuantumSavory/QuantumSavory.jl #48: Current release broken

Abhishek-1Bhatt, 2023-10-09:
I was trying to install `QuantumSavory` in a Pluto notebook and noticed that the current release `v0.3.0` won't precompile because its missing the `consistent_representation` method defined [here](https://github.com/QuantumSavory/QuantumSavory.jl/blob/8710dce6ae58982aff04bc078ee28992cb5336d6/src/baseops/initialize.jl#L39-L42) which is causing [this](https://github.com/QuantumSavory/QuantumSavory.jl/blob/8710dce6ae58982aff04bc078ee28992cb5336d6/src/precompile.jl#L22) line in src/precompile.jl to error

```julia
ERROR: LoadError: MethodError: no method matching consistent_representation(::Vector{QuantumSavory.Register}, ::Vector{Int64}, ::QuantumSymbolics.XBasisState)

Closest candidates are:
  consistent_representation(::Any, ::Any)
   @ QuantumSymbolics C:\Users\Abhishek Bhatt\.julia\packages\QuantumSymbolics\QFyHe\src\QSymbolicsBase\express.jl:35

Stacktrace:
  [1] initialize!(r::Vector{QuantumSavory.Register}, i::Vector{Int64}, state::QuantumSymbolics.XBasisState; time::Float64)
    @ QuantumSavory C:\Users\Abhishek Bhatt\.julia\dev\QuantumSavory\src\baseops\initialize.jl:37
  [2] initialize!(reg::QuantumSavory.Register, i::Int64, state::QuantumSymbolics.XBasisState; time::Float64)
    @ QuantumSavory C:\Users\Abhishek Bhatt\.julia\dev\QuantumSavory\src\baseops\initialize.jl:35
  [3] initialize!(r::QuantumSavory.RegRef, state::QuantumSymbolics.XBasisState; time::Float64)
    @ QuantumSavory C:\Users\Abhishek Bhatt\.julia\dev\QuantumSavory\src\baseops\initialize.jl:36
  [4] macro expansion
    @ C:\Users\Abhishek Bhatt\.julia\dev\QuantumSavory\src\precompile.jl:22 [inlined]
  [5] macro expansion
    @ C:\Users\Abhishek Bhatt\.julia\packages\PrecompileTools\kmH5L\src\workloads.jl:78 [inlined]
  [6] macro expansion
    @ C:\Users\Abhishek Bhatt\.julia\dev\QuantumSavory\src\precompile.jl:6 [inlined]
  [7] top-level scope
    @ C:\Users\Abhishek Bhatt\.julia\packages\PrecompileTools\kmH5L\src\workloads.jl:140
  [8] include(mod::Module, _path::String)
    @ Base .\Base.jl:457
  [9] include(x::String)
    @ QuantumSavory C:\Users\Abhishek Bhatt\.julia\dev\QuantumSavory\src\QuantumSavory.jl:1
 [10] top-level scope
    @ C:\Users\Abhishek Bhatt\.julia\dev\QuantumSavory\src\QuantumSavory.jl:337
 [11] include
    @ .\Base.jl:457 [inlined]
 [12] include_package_for_output(pkg::Base.PkgId, input::String, depot_path::Vector{String}, dl_load_path::Vector{String}, load_path::Vector{String}, concrete_deps::Vector{Pair{Base.PkgId, UInt128}}, source::String)
    @ Base .\loading.jl:2049
 [13] top-level scope
    @ stdin:3
in expression starting at C:\Users\Abhishek Bhatt\.julia\dev\QuantumSavory\src\precompile.jl:3
in expression starting at C:\Users\Abhishek Bhatt\.julia\dev\QuantumSavory\src\QuantumSavory.jl:1
in expression starting at stdin:3
```

Krastanov, 2023-10-17:
Weird, I wonder what happened here. The CI worked at the time and master works. I will rerun some tests and make a release

Abhishek-1Bhatt, 2023-10-17:
fixed in v0.3.1

closed 2023-10-17 by Abhishek-1Bhatt