QuantumSavory/QuantumSavory.jl #129: Tensor Networks Backend [$200]

Krastanov, 2024-07-09:
<details>
<summary><strong>Bug bounty logistic details</strong> (click to expand)</summary>

To claim exclusive time to work on this bounty either post a comment here or message [skrastanov@umass.edu](mailto:skrastanov@umass.edu) with:

- your name
- github username
- **(optional)** a brief list of previous pertinent projects you have engaged in

Currently the project is claimed by `no one` until `...`.

If you want to, you can work on this project without making a claim, however claims are encouraged to give you and other contributors peace of mind. Whoever has made a claim takes precedence when solutions are considered.

You can always propose your own funded project, if you would like to contribute something of value that is not yet covered by an official bounty.
</details>

# Project: "Tensor Networks Backend" [$200]

This project depends on first completing https://github.com/QuantumSavory/QuantumSymbolics.jl/issues/63

This simulation library can easily work with state vectors and stabilizer tableaux, thanks to the existence of QuantumSymbolics backends for them. **To complete this project the contributor will need to plug in a tensor network backend, so that simulations can be performed using tensor network numerical representations.** Some of the currently existing examples already perform simulations using both stabilizer states and state vectors. These examples will need to be updated to include tensor network simulations as well. A new example of a simulation that would be too expensive to run without tensor networks would be necessary as part of this contribution, in order to showcase and document the new functionality.

**Required skills**: Familiarity with tensor networks and with the QuantumSymbolics package.

**Reviewer**: Stefan Krastanov

**Duration**: Can be reserved for 2 months and potentially extended.

**Publication**: In the next 2 years we plan to release a paper in a selective journal about this software. Contributing to this issue would deserve a co-authorship status on such a paper (if the contributor so desires)

#### Payout procedure:

The Funding for these bounties comes from the National Science Foundation and from the NSF Center for Quantum Networks. The payouts are managed by the NumFOCUS foundation and processed in bulk once every two months. If you live in a country in which NumFOCUS can make payments, you can participate in this bounty program.

[Click here for more details about the bug bounty program.](https://github.com/QuantumSavory/.github/blob/main/BUG_BOUNTIES.md)

limitless-100, 2025-04-04:
@Krastanov Please, is this bounty still open to new contributors?

Krastanov, 2025-04-06:
Hi, Adesina! Apologies for the delayed response. This issue depends on the linked issue in QuantumSymbolics, so it would not be doable to start it before the one in QuantumSymbolics is finished. I will contact the person that is working on the QuantumSymbolics side to check how they are progressing

arnavk23, 2025-11-30:
@Krastanov can I taken on this bounty issue?