QuantumSavory/QuantumSavory.jl #130: Improve static analysis tests with JET [$200]

Krastanov, 2024-07-09:
<details>
<summary><strong>Bug bounty logistic details</strong> (click to expand)</summary>

To claim exclusive time to work on this bounty either post a comment here or message [skrastanov@umass.edu](mailto:skrastanov@umass.edu) with:

- your name
- github username
- **(optional)** a brief list of previous pertinent projects you have engaged in

Currently the project is claimed by `no one` until `...`.

If you want to, you can work on this project without making a claim, however claims are encouraged to give you and other contributors peace of mind. Whoever has made a claim takes precedence when solutions are considered.

You can always propose your own funded project, if you would like to contribute something of value that is not yet covered by an official bounty.
</details>

# Project: "Improve static analysis tests with JET" [$200]

We use JET for static analysis of the codebase. **To claim this bounty, fix 2/3 of the reported warnings**. Some of them are false positives that can be removed with extra type annotation. Some of them are due to dependencies, in which case you can simply blacklist the dependency (as already done). See the `test_jet.jl` test file and the JET CI run for the current state of affairs.

**Required skills**: General Julia and JET skills.

**Reviewer**: Stefan Krastanov

**Duration**: 1 month

#### Payout procedure:

The Funding for these bounties comes from the National Science Foundation and from the NSF Center for Quantum Networks. The payouts are managed by the NumFOCUS foundation and processed in bulk once every two months. If you live in a country in which NumFOCUS can make payments, you can participate in this bounty program.

[Click here for more details about the bug bounty program.](https://github.com/QuantumSavory/.github/blob/main/BUG_BOUNTIES.md)

mawutory, 2024-07-20:
hi @Krastanov, please i would like to take this on as a first issue.

Krastanov, 2024-07-21:
Hi, @mawutory ! Do you have past experience with Julia and the JET static analysis tool? While this issue should be relatively easy for someone with experience, it might be a bit hard jumping in if you are a novice, and starting with some of the other bug bounties might be easier.

mawutory, 2024-07-21:
hello @Krastanov, thanks for the warm introduction. i do have quite some experience in python but i'm fairly new to Julia. however, i realized that Julia syntax is quite similar to that of python so i do not think the language will be a problem for me but i do concede that i need time to understand the codebase and high-level computing concepts in general.

please i'd appreciate your advice on how think i could get started. what are the issues that you think could be a good first issue?

Krastanov, 2024-07-22:
Here are some notes on Julia that might be useful:

- https://lab.krastanov.org/sysadminonboarding/
- https://modernjuliaworkflows.github.io/

And here is the link from where you can see all bug bounties: https://github.com/QuantumSavory/.github/blob/main/BUG_BOUNTIES.md

You can also search for the tag "good first issue".

mawutory, 2024-07-22:
thanks @Krastanov

lgoettgens, 2024-09-08:
@Krastanov I would like to give this a shot. I am already a bit somewhat familiar with fixing JET related issue from my day job, see e.g. https://github.com/Nemocas/Nemo.jl/pull/1762.

Currently, there are 146 issues reported by JET:
https://github.com/QuantumSavory/QuantumSavory.jl/blob/88d64ca748efb39c4b6755f63df711e7715c7dca/test/test_jet.jl#L41
2/3 of them is 97, so I would try to decrease the number by that amount.

Krastanov, 2024-09-09:
That would be great! Thank you for your interest! I will mark this as reserved.

Krastanov, 2024-09-09:
If there is a problem with some API decision or something in a dependency, do not hesitate to post and ask for suggestions or help.

closed 2024-09-14 by Krastanov