QuantumSavory/QuantumSavory.jl #131: More thorough benchmarks [$200]

Krastanov, 2024-07-09:
<details>
<summary><strong>Bug bounty logistic details</strong> (click to expand)</summary>

To claim exclusive time to work on this bounty either post a comment here or message [skrastanov@umass.edu](mailto:skrastanov@umass.edu) with:

- your name
- github username
- **(optional)** a brief list of previous pertinent projects you have engaged in

Currently the project is claimed by `no one` until `...`.

If you want to, you can work on this project without making a claim, however claims are encouraged to give you and other contributors peace of mind. Whoever has made a claim takes precedence when solutions are considered.

You can always propose your own funded project, if you would like to contribute something of value that is not yet covered by an official bounty.
</details>

# Project: "More thorough benchmarks" [$200]

We have a small benchmark suite already implemented, which is executed as part of our CI runs. It is defined in the `benchmark` folder and reported for each pull request. **We would like to expand this benchmark suite** to include many more facets of this library. E.g. basic register operations using a variety of backends; queries and tags and locks on registers and channels; time to import; time to run examples. The new benchmarks should be legible, easy to follow, and organized by topic. Most of them should be microbenchmarks testing only one concept, but a few holistic benchmarks would make sense as well.

**Required skills**: Generic Julia skills.

**Reviewer**: Stefan Krastanov

**Duration**: 1 month

#### Payout procedure:

The Funding for these bounties comes from the National Science Foundation and from the NSF Center for Quantum Networks. The payouts are managed by the NumFOCUS foundation and processed in bulk once every two months. If you live in a country in which NumFOCUS can make payments, you can participate in this bounty program.

[Click here for more details about the bug bounty program.](https://github.com/QuantumSavory/.github/blob/main/BUG_BOUNTIES.md)

coderight1, 2024-07-10:
Hi @Krastanov , I have recently started learning Julia , and willing to work on this.

Krastanov, 2024-07-10:
Please feel free to go ahead with it. Before I reserve it for you though, let's discuss what you plan to add in this comment thread. While this is a good issue for a first-time developer, it is not an easy issue for someone who has not written any user code with this library. I would suggest trying to run a few examples and creating a simulation of something interesting first. The benchmarks are supposed to have a very thorough coverage.

coderight1, 2024-07-11:
Hi @Krastanov,

Thank you for the opportunity. I appreciate your advice and would like to proceed with the following plan for expanding the benchmark suite:

Basic Register Operations:
Add benchmarks for creating, updating, and deleting registers using various backends.

Queries, Tags, and Locks:
Benchmark the time for queries, adding/removing tags, and locking/unlocking operations on registers and channels.

Import Time:
Measure the time it takes to import the library in different environments.

Example Run Time:
Benchmark the execution time of existing examples.
Create and benchmark new examples to cover additional scenarios.

To keep things organized, I plan to create different files for each facet of the library, such as basic register operations, queries and tags, locks on registers and channels, etc. This way, the benchmarks will be easier to manage and understand.

Please let me know if there are any specific areas you would like me to prioritize or any additional details I should consider.

Krastanov, 2024-07-12:
Sounds good! A few suggestions:

- Check out the benchmarks setup of QuantumClifford.jl for reference
- Be careful with the eval and samples keywords: in some situations, especially if you are modifying/deleting things, you have to have eval=1, otherwise you get very misleading results
- Make sure the code is relatively easy to execute interactively benchmark by benchmark (i.e. factor out neatly potential setup/prepartion routines so a dev who wants to experiment knows what they need to load in their REPL)
- Have benchmarks both for large number of entities and for small number of entities (e.g. both large networks and small networks, large registers and registers with only a few slots, queries on large number of prerecorded tags and just a few tags)
- Be careful with setting things up that might require the discrete event simulator. There are many functions that just "set up an event" and the event does not run until you execute the simulator. Particularly around locks. I would suggest grouping this separately from the tag/query topic.
- For import time, it might be a bit awkward, because the benchmark will basically have to launch a separate julia instance. That also probably should have a limited number of eval and samples. Make sure that you actually wait on the newly started process (otherwise you might just launch it in the background and get a report that it is incredibly fast)
- Testing duration of examples might also be difficult (this is actually why this project is 200$ instead of 50$). No need to test all of them, especially ones that contain plots are probably a bad idea. You can see in the test_examples how we run them for tests, that would probably be a good reference (and it will make it more obvious which ones involve plotting and should be skipped). These should probably have just one eval and one sample given how incredibly slow they can be. It would be important to run these in a separate module (e.g. the way they are done in the tests). I suspect you will need something like `eval(Module(), :(include("example file")))` to run this benchmark.
- Make sure that you also run each benchmark you define interactively (by changing `@benchmarkable` to `@benchmark`) just to see what would be the dev experience of someone who is running a single benchmark manually to explore why there is a slowdown in some future PR
- avoid using loops to generate benchmark instances (e.g. if you are just changing a single parameter). Sure, that would lead to more repetitive code and duplication, but it will help when in the future someone wants to play with a specific benchmark

I will make this reserved for 1month, but we can extend if necessary. Thank you for taking this on.

Krastanov, 2024-08-14:
Hi, @coderight1! I have not heard from you, let me know if you are still interested in this. I would like to open it up soon so other have a chance to engage with it.

aaryansinhaa, 2025-08-12:
Hi @Krastanov ,

I am new to Julia, and I would like to take this on as a challenge.
I checked the `benchmark.jl` file and reviewed the benchmark setup of `QuantumClifford.jl` for reference.

Before starting, I plan to spend some time learning the `BenchmarkTools.jl` package in detail to ensure accurate and consistent results.

I saw the features proposed earlier by @coderight1  and would like to implement those, including:
- Micro-benchmarks for basic register operations, queries, tags, and locks
- Import time measurements
- Example run time measurements and expansions

In addition, I plan to split the current monolithic `benchmark.jl` into smaller, focused files for better organization:
- `bench_register_operations.jl` → create, update, delete registers on different back-ends.
- `bench_queries_tags.jl` → querying, adding/removing tags, and related operations
- `bench_locks.jl` → locking/unlocking registers and channels
- `bench_import.jl` → measuring import time across environments
- `bench_examples.jl` → runtime measurement of current and newly added usage examples

I plan to incrementally add these benchmarks while optimizing the setup to reduce noise in the measurements. I understand that benchmarks need to have a very thorough coverage, and hence I would really appreciate if you could provide me with some pointers and documentations to refer, before I start to work on this bounty.

Could you also please confirm if the bounty is still open?

aaryansinhaa, 2025-08-13:
Hey @Krastanov ,

I spent some time with the codebase, and I feel that I will be able to complete this task within 1-1.5 month. 
Here is the plan that I am proposing for the next two weeks:

## Week 1:
- Understand the Codebase.
- Understand the working and examples in Quantum Savory, fully understand the concepts, structures and functions in the library.
- Study the benchmarks of Quantum Clifford.
-  I'll carefully split the existing `benchmark.jl` into the new, modular files, ensuring that the `BenchmarkGroup` structure remains compatible with other Julia benchmarking tools. This will be my primary focus.

## Week 2: 
- Make the benchmarks for register related operations in `bench_register_operations.jl`.
- Ensure that the benchmarks don't break, so that the new benchmarks are accurate and do not produce misleading results, especially when dealing with state-modifying operations.
- Focus on micro benchmarking for register operations.

To allow me to focus entirely on the task, could you please reserve this bounty for me?
I'd also like to ask for your permission to reach out to you on the Julia Slack. It would allow for a more direct and efficient conversation if I have any quick questions during the development process.

Thank you for this opportunity!

Krastanov, 2025-08-13:
Sounds great! Marking it as reserved.

If you use LLMs please mark that in your commit messages and in the opening message for PRs. These tools are great, but it is really important to know when they are used as it informs how the review is done.

It would be really helpful to submit many small PRs instead of one large one -- a large PR might never get reviewed.

floraiNx, 2026-05-27:
Hi Stefan, I took a closer look at the current benchmark suite and the recent closed benchmark PRs before opening anything. I also ran the benchmark setup locally with Julia 1.12 and smoke-tested a small QTCP benchmark prototype.

To avoid duplicating the register/protocol/import benchmark attempts, I would like to start with a narrow QTCP-focused slice: benchmark a small complete QTCP chain flow using the existing `EndNodeController`, `NetworkNodeController`, and `LinkController`, with assertions that the expected `QTCPPairBegin` and `QTCPPairEnd` tags are delivered.

The local prototype uses the existing linear `RegisterNet([Register(...)])` setup, so it does not require adding a direct `Graphs` dependency to the benchmark project. I would keep this as a small PR, likely under a `protocolzoo/qtcp` benchmark group, rather than trying to cover all examples or import time in one patch.

Would that be a useful first PR toward this issue? If I open it, I will include local run notes and disclose any AI assistance in the PR body as requested earlier in the thread.

CookUpSunny, 2026-05-28:
Opened #439 with a focused `register_locks` benchmark slice for #131.

This adds coverage for direct `RegRef` lock checks, single-slot lock contention, multi-slot acquisition, and contended `nongreedymultilock` / deterministic `spinlock` scenarios. I kept it scoped to register lock behavior to avoid duplicating the existing tag/query benchmark work.

Local validation is included in the PR body.

Thanhdn1984, 2026-05-30:
Hi, I would like to claim this bounty for Thanhdn1984.

GitHub username: @Thanhdn1984
Relevant background: automation/tooling and benchmark-oriented repo work via AI-assisted development; I can inspect the existing `benchmark/` suite, add topic-organized Julia microbenchmarks, keep them legible, and submit a PR for review.

Planned scope: expand benchmarks for register operations, queries/tags/locks, import timing, examples/runtime paths, plus a small set of holistic benchmarks, matching the existing CI benchmark style.

I will wait for maintainer acknowledgement before doing substantial implementation work.

yanyishuai, 2026-06-29:
Claiming this bounty — preparing expanded benchmark suite PR.