QuantumSavory/QuantumSavory.jl #132: Improve Makie visualization capabilities [$200]

Krastanov, 2024-07-09:
<details>
<summary><strong>Bug bounty logistic details</strong> (click to expand)</summary>

To claim exclusive time to work on this bounty either post a comment here or message [skrastanov@umass.edu](mailto:skrastanov@umass.edu) with:

- your name
- github username
- **(optional)** a brief list of previous pertinent projects you have engaged in

Currently the project is claimed by `no one` until `...`.

If you want to, you can work on this project without making a claim, however claims are encouraged to give you and other contributors peace of mind. Whoever has made a claim takes precedence when solutions are considered.

You can always propose your own funded project, if you would like to contribute something of value that is not yet covered by an official bounty.
</details>

# Project: Improve Makie visualization capabilities [$200]

We use Makie for visualizing the state of a simulated network. We can visualize the state of registers, the content of these registers, some tag information, etc. There is much more information that can be tracked visually though. This project is about fleshing out these visualization features.
[This list of issues can be a useful guide to features that need to be implemented](https://github.com/QuantumSavory/QuantumSavory.jl/issues?q=is%3Aissue+is%3Aopen+label%3Avisualization), but you should feel free to suggest deviations from these guidelines if you have different ideas -- however the scope of the project should be set at the start of the project, e.g. by discussing it in this thread.

**Required skills**: Understanding of the Makie plotting library

**Reviewer**: Stefan Krastanov

**Duration**: 1 month

#### Payout procedure:

The Funding for these bounties comes from the National Science Foundation and from the NSF Center for Quantum Networks. The payouts are managed by the NumFOCUS foundation and processed in bulk once every two months. If you live in a country in which NumFOCUS can make payments, you can participate in this bounty program.

[Click here for more details about the bug bounty program.](https://github.com/QuantumSavory/.github/blob/main/BUG_BOUNTIES.md)

Arthurescc, 2026-05-15:
Following the bounty logistics template:

- name: Doyele
- github username: Arthurescc
- relevant background: submitted PR #411 for the #98 state-tooltip part of this Makie visualization bounty; comfortable iterating on Julia/Makie plotting code and tests based on maintainer feedback.

I understand this is a narrow first slice rather than a full solution to every visualization sub-issue under #132.

Krastanov, 2026-05-16:
@oliverkimlamlam , if you have not worked with Julia before (what I assume given the fact you can not yet run tests on your system), this bounty would not be a good place to start.

@EugeniSegarra , of all of our software projects, QuantumSavory.jl will probably indeed be the most approachable given your background. How to proceed probably would depend on what you want to get out of this. If the focus is more on the Julia ecosystem as a whole, probably focusing on these visualization issues is a good place to start. But for anything else, you probably would need to build up some background in quantum information science, which would require going through a course or a textbook (plenty of good options online, e.g. youtube video lectures). Plenty of PRs were closed without review because they were spam -- that does not mean the topic they were addressing is not in scope under this bounty.

oliverkimlamlam, 2026-05-16:
Thanks Stefan, understood. I’ll step back from this bounty for now and won’t open a PR for this scope. I’ll first get a local Julia/Makie environment working and spend time learning the project before considering future contributions.

DariusJDebugs, 2026-08-16:
Hi Stefan,

I'm interested in working on the $200 Makie visualization bounty and would like to claim it if it is still available.

Before beginning, could you please confirm:

1. Is the bounty still funded and available?
2. What specific visualization features should be included?
3. What exact acceptance criteria or tests must the solution satisfy?
4. How long would my exclusive claim remain active?
5. Is a U.S.-based contributor eligible for payment through NumFOCUS?
6. Are AI-assisted contributions allowed if I personally review, test, and take responsibility for the completed work?

Once the scope and acceptance requirements are confirmed, I can evaluate the implementation and begin promptly.

Thank you!

bilhokista, 2026-09-12:
Hi @Krastanov — claiming this bounty as well, scoped to a specific item from the visualization list.

- **Name:** Dian Bilhokista
- **GitHub:** @bilhokista

### Scope: #96 — show message buffers in the hover tooltips

I picked the narrowest item on the list so the scope is unambiguous. Current state, as I read it:

`ext/QuantumSavoryMakie/QuantumSavoryMakie.jl:238` wires the slot scatter plot to `get_slots_vis_string`, which renders register index, slot index and the slot's tags. The register rectangle has `inspectable[] = false` with a TODO noting the `Poly` plot does not inspect properly, so the slot markers are the only working hover target today.

`messagebuffer(net, node)` returns a `MessageBuffer{Tag}` whose `.buffer` holds `(src, tag)` entries, so the data is already there — the tooltip just has no access to the network, since `get_slots_vis_string` only receives `(register, registeridx, slot)`.

### Proposed change

1. Give the slot tooltip access to the network (pass the network observable into the `inspector_label` closure rather than widening the backref tuple, so nothing else has to change).
2. Append a pending-messages section to the slot tooltip, showing the count and the `(src, tag)` entries for that node, truncated to a few lines so a busy buffer does not produce an unreadable tooltip.
3. A test under `test/plotting/` that puts a known tag into a node's buffer and asserts the rendered string contains it.

Two questions before I write it:

1. **Placement.** The message buffer belongs to the node, not the slot, so showing it on every slot of a register repeats it. Would you rather I (a) show it on each slot anyway since those are the only inspectable markers, or (b) first fix the register `Poly` inspectability so the buffer can live on the register rectangle where it belongs? Option (b) is the cleaner result but a wider change, and it touches the TODO at line 233.
2. **Truncation.** Is there a preferred cap for how many buffer entries to list before eliding? I would default to five plus a "and N more" line unless you have a convention.

Happy to also take #97 or #98 afterwards if this one lands well — though I read your note on #98 that the entangled-state case is still unsolved, so I would leave that one alone for now.

omnesandre, 2026-09-14:
I'd like to work on this bounty. Is it still available, and may I claim/reserve it before I begin?