QuantumSavory/QuantumSavory.jl #134: Entanglement tracker protocol with UUIDs assigned to each Bell pair [$1600]

Krastanov, 2024-07-09:
<details>
<summary><strong>Bug bounty logistic details</strong> (click to expand)</summary>

To claim exclusive time to work on this bounty either post a comment here or message [skrastanov@umass.edu](mailto:skrastanov@umass.edu) with:

- your name
- github username
- **(optional)** a brief list of previous pertinent projects you have engaged in

Currently the project is claimed by `no one` until `...`.

If you want to, you can work on this project without making a claim, however claims are encouraged to give you and other contributors peace of mind. Whoever has made a claim takes precedence when solutions are considered.

You can always propose your own funded project, if you would like to contribute something of value that is not yet covered by an official bounty.
</details>

# Project: Entanglement tracker protocol with UUIDs assigned to each Bell pair [$1600]

An important feature for any simulation involving non-trivial networking is tracking entanglement swaps. Consider a chain of repeaters between nodes A and B. Only "link-level" raw entanglement can be generated, entangling only nearest neighbors. Entanglement swaps are necessary at each repeater in order to take the many "link-level" pairs and turn them into one pair between A and B. However, whenever a swap happens, classical messages need to be sent through the network, alerting the other end of a pair that has undergone a swap that they are now entangled to a completely different node. One such set of protocols is currently implemented in `EntanglementTracker` (and `EntanglerProt` and `SwapperProt`), using tags like `EntanglementUpdate` and `EntanglementHistory`. This protocol is very "stateful" and requires detailed history to be kept.

This bounty is about implementing an alternative protocol, one in which each pair is assigned a UUID and entanglement tracking and updates use these IDs instead of trying to keep as detailed of a history.

Detailed documentation, visualizations, examples, and comparisons to the already available protocols also would need to be provided as part of this bounty.

**Required skills**: Understanding of entanglement swapping, and concurrent programming with the ConcurrentSim.jl and ResumableFunctions.jl libraries.

**Reviewer**: Stefan Krastanov

**Duration**: 2 months but can easily be extended

**Publication**: In the next 2 years we plan to release a paper in a selective journal about this software. Contributing to this issue would deserve a co-authorship status on such a paper (if the contributor so desires)

#### Payout procedure:

The Funding for these bounties comes from the National Science Foundation and from the NSF Center for Quantum Networks. The payouts are managed by the NumFOCUS foundation and processed in bulk once every two months. If you live in a country in which NumFOCUS can make payments, you can participate in this bounty program.

[Click here for more details about the bug bounty program.](https://github.com/QuantumSavory/.github/blob/main/BUG_BOUNTIES.md)

pooryousefshahrooz, 2024-08-13:
Name: Shahrooz Pouryousef
github username:pooryousefshahrooz

riteshbhirud, 2025-03-11:
@Krastanov sent a message over Zulip to check in regarding this

vivek-kumar9696, 2025-07-23:
Hi @Krastanov ,
Is this still valid? I have considerable experience with research projects of this kind and would love to contribute.

Krastanov, 2025-08-12:
Hi, @vivek-kumar9696 , apologies for the slow response. The project is indeed still open. Let me know if you would like to discuss a potential bounty on this.