QuantumSavory/QuantumSavory.jl #137: Implement a networking protocol primitive [$400]

Krastanov, 2024-08-01:
<details>
<summary><strong>Bug bounty logistic details</strong> (click to expand)</summary>

To claim exclusive time to work on this bounty either post a comment here or message [skrastanov@umass.edu](mailto:skrastanov@umass.edu) with:

- your name
- github username
- **(optional)** a brief list of previous pertinent projects you have engaged in

Currently the project is claimed by `no one` until `...`.

If you want to, you can work on this project without making a claim, however claims are encouraged to give you and other contributors peace of mind. Whoever has made a claim takes precedence when solutions are considered.

You can always propose your own funded project, if you would like to contribute something of value that is not yet covered by an official bounty.
</details>

# Implement a networking protocol primitive [$400]

There are many interesting protocols and networking primitives yet to be added to this library. To claim this bounty propose and then implement a protocol for the ProtocolZoo submodule. Multiple people can claim this bounty (repeatedly) for new implementations. For substantial development efforts bigger bounties can be approved.

Tests and documentation have to be part of such submissions.

**Required skills**: Deep knowledge of the protocol being implemented and rough understanding of networking simulations with this library.

**Reviewer**: Stefan Krastanov

**Duration**: 2 months

**Publication**: In the next 2 years we plan to release a paper in a selective journal about this software. Contributing to this issue would deserve a co-authorship status on such a paper (if the contributor so desires)

#### Payout procedure:

The Funding for these bounties comes from the National Science Foundation and from the NSF Center for Quantum Networks. The payouts are managed by the NumFOCUS foundation and processed in bulk once every two months. If you live in a country in which NumFOCUS can make payments, you can participate in this bounty program.

[Click here for more details about the bug bounty program.](https://github.com/QuantumSavory/.github/blob/main/BUG_BOUNTIES.md)

idriss-hamadi, 2024-09-06:
Hello @Krastanov,

I’d like to tackle this issue. Could you please let me know if there is any documentation available for the protocols that have already been implemented? so i can check it and see how would you like it to be structured, Additionally, if there is any specific protocols that you may want to be implemented firs? I’d appreciate the information

Krastanov, 2024-09-11:
Hi, @idriss-hamadi Pardon the slow response. I will respond to your recent email and when we have a specific protocol we can make a dedicated issue to track the bounty.

sagnikpal2004, 2024-09-19:
Hi @Krastanov 
@mantri-prateek and I would like to work on this issue to implement a two-way network protocol

Krastanov, 2024-09-23:
Sagnik, Prateek, thanks for the interest! I will message you over email so we can discuss details of what you have in mind.

Krastanov, 2024-09-26:
For public tracking: @mantri-prateek and @sagnikpal2004 are working on an instance of this bounty, related to swaps and distillation in a repeater chain. This bounty itself is left unreserved as it can be claimed repeatedly, e.g. by other community members contributing unrelated protocols and examples.

nenadilic84, 2026-02-13:
I would like to propose implementing a **BBM92 entanglement-based quantum key distribution** protocol for the ProtocolZoo.

**Protocol**: BBM92 (Bennett, Brassard, Mermin 1992) is an entanglement-based QKD protocol where two parties share Bell pairs and each independently measures their qubit in a randomly chosen basis (Z or X). After publicly comparing basis choices, they keep only the matching-basis outcomes as secret key bits. A subset of the sifted key is sacrificed to estimate the quantum bit error rate (QBER).

**Why it fits QuantumSavory**: BBM92 is the canonical application-layer protocol for entanglement distribution networks — it turns shared Bell pairs into cryptographic key material. It naturally sits on top of the existing EntanglerProt → SwapperProt → EntanglementTracker pipeline and complements EntanglementConsumer by adding basis randomization, key sifting, and QBER estimation. It demonstrates the practical utility of the entire simulation stack.

**Deliverables**: `BBM92Prot` protocol struct with helper functions (`sifted_key`, `qber_estimate`, `keyrate`), comprehensive tests over direct links and repeater chains, interactive GLMakie/WGLMakie visualization showing real-time key generation rate, QBER, and basis sifting statistics over a repeater chain, HTML show method, and CHANGELOG entry.

I have an implementation ready — will open a PR shortly.

franklincg, 2026-08-21:
Hi @Krastanov — I'd like to claim one instance of this repeatable $400 bounty with a specific ProtocolZoo primitive.

### Proposal: DEJMPS entanglement purification protocol

I propose implementing a focused DEJMPS purification primitive for two noisy Bell pairs. This would be separate from the existing swap/distillation repeater-chain work by providing a reusable ProtocolZoo component with explicit protocol state and testable success/failure behavior.

Planned scope:
- a DEJMPS protocol component integrated with the existing register/tag/locking conventions;
- selection/locking of two eligible entangled pairs and reciprocal metadata validation;
- modeled local bilateral operations, measurements, classical comparison, and keep/discard cleanup;
- deterministic tests for success, rejection, stale-query/race, and cleanup paths;
- documentation + a minimal usage example demonstrating fidelity improvement across repeated trials.

If DEJMPS is an acceptable distinct instance of #137, I can proceed with the implementation.