QuantumSavory/QuantumSavory.jl #139: A (web)GUI for drawing out a network, assigning protocols, and running simulations with some introspection [$3000]

Krastanov, 2024-08-01:
<details>
<summary><strong>Bug bounty logistic details</strong> (click to expand)</summary>

To claim exclusive time to work on this bounty either post a comment here or message [skrastanov@umass.edu](mailto:skrastanov@umass.edu) with:

- your name
- github username
- **(optional)** a brief list of previous pertinent projects you have engaged in

Currently the project is claimed by `no one` until `...`.

If you want to, you can work on this project without making a claim, however claims are encouraged to give you and other contributors peace of mind. Whoever has made a claim takes precedence when solutions are considered.

You can always propose your own funded project, if you would like to contribute something of value that is not yet covered by an official bounty.
</details>

# A (web)GUI for drawing out a network, assigning protocols, and running simulations with some introspection [$3000]

We already have some plotting and visualization tools, however we are interested in setting up a semi-independent GUI for drawing a network (a graph of connected quantum registers), assigning various protocols to each node, and running the simulation, while being able to read out or plot some instrumentation data about the network. This bounty is for an initial prototype of such a GUI.

Given that this is a rather large project, we are also open to hiring contractors that want to work on a much larger scale for this goal, which of course would include much higher payment as the project progresses.

**Required skills**: Probably significant JS skills for designing the frontend. Understanding of the needs of network scientists.

**Reviewer**: Stefan Krastanov

**Duration**: 4 months or more

**Publication**: In the next 2 years we plan to release a paper in a selective journal about this software. Contributing to this issue would deserve a co-authorship status on such a paper (if the contributor so desires)

#### Payout procedure:

The Funding for these bounties comes from the National Science Foundation and from the NSF Center for Quantum Networks. The payouts are managed by the NumFOCUS foundation and processed in bulk once every two months. If you live in a country in which NumFOCUS can make payments, you can participate in this bounty program.

[Click here for more details about the bug bounty program.](https://github.com/QuantumSavory/.github/blob/main/BUG_BOUNTIES.md)

Myestery, 2024-08-01:
Hello @Krastanov 
My name is Johnpaul 
I am interested in this bounty, I specialise in building web clients for products like this.
If you don't mind, I can send a proposal with a proposed design for the platform when all clients needs are well understood

Krastanov, 2024-08-01:
Hi, Chiwetelu, and thanks for your interest! This is a pretty big project and will require Julia knowledge in order to hook into the underlying simulation package, in addition to GUI/Web programming skills. I would be very happy to discuss your initial proposal and help you polish it, however for bounty proposals over 800$ we require some proof of capabilities, kinda like with Google Summer of Code -- e.g. can you submit a pull request addressing one of the smaller bounties or one of the "good first issues" as a qualification step? You can do this in parallel with discussing and polishing your proposal if you wish.

lluvimaa, 2025-05-13:
Is this issue still relevant? @Krastanov

Krastanov, 2025-05-13:
Thanks for flagging it. No, we have someone internally who has started working on this. We will probably reopen the issue once we have the first prototype ready and have specific improvements planned for it.

closed 2025-05-13 by Krastanov

lluvimaa, 2025-05-13:
There are bounty issues in other repos opened a while ago, anything worth looking at (not closed/taken/outdated)? @Krastanov

Krastanov, 2025-05-13:
Most of them are still open and later in may I will be doing a refresh of them. Feel free to email me at stefan@krastanov.org if you want to discuss any larger related paid projects you might be interested in, even if they are no existing bounties for them yet.