QuantumSavory/QuantumSavory.jl #211: Display state in order of its state indices

Luisenden, 2025-04-08:
@KDGoodenough and I thought this might be interesting: 
In our current project, we use a quantum switch to generate a target graph state and then compare the final output to the reference state to ensure they match (see code example below); however, we noticed that the stabilizer representation of the reference state (`refstate`) differs from that of the final state (`clients[1].staterefs[1].state[]`). More specifically, even though they share the same underlying graph structure, their vertex labelling is off.

What happened?
The discrepancy arises because the final state is stored and displayed **in the order of the client registers**, rather than following the internal `stateindices`. This shifts the vertex labeling and creates what appears to be a “jumbled” output (for instance, `[clients[i].stateindices[1] for i in 1:4] = [[1], [2], [4], [3]]`). As a result, although the two states are identical in a physical sense, their adjacencies and therefore the sum of stabilizers look different. 

**Is there a way to display a state in the order of its state indices?**
We for now use a function `order_state!` that physically swaps the qubits into the expected order. 

```julia
using QuantumSavory
using QuantumClifford: Stabilizer, canonicalize!, sSWAP, AbstractStabilizer
using QuantumOpticsBase
using Graphs

function order_state!(state::AbstractStabilizer, current_order::Vector{Int})
    # Loop over each index 
    for i in 1:length(current_order)
        # If the qubit at position i isn't i, swap it with wherever qubit i lives
        while current_order[i] != i
            @info "current order $(current_order)"
            # Find which position holds the qubit i
            correct_index = findfirst(==(i), current_order)

            # Swap the qubits physically
            apply!(state, sSWAP(current_order[i], current_order[correct_index]); phases=true)
            current_order[i], current_order[correct_index] = current_order[correct_index], current_order[i]
            @info "swapped $((i,correct_index)) to get $(collect(edges(graphstate(state)[1])))"
        end
    end
    @info "current order $(current_order)"
end

# Toy version of canonical protocol
n = 4 # number of qubits

# Define graph
g = random_regular_graph(n, 2, seed=2)
refstate = Stabilizer(g)

# Set up registers
switch = Register(fill(Qubit(), n), fill(CliffordRepr(), n)) 
clients = [Register(1, CliffordRepr()) for _ in 1:n]

# Initialize Bell pairs
bell = StabilizerState("XX ZZ")
for i in 1:n
    initialize!([clients[i][1], switch[i]], bell)
end

# Apply CZ gates at the switch according to graph
measured_out = []
for i in 1:n
    neighs = copy(neighbors(g, i))
    for j in neighs
        apply!([switch[i], switch[j]], ZCZ)
        rem_edge!(g, i, j)
    end

    # Measure qubit in X basis and apply Z correction
    xmeas = project_traceout!(switch[i], σˣ)
    push!(measured_out, i)
    if xmeas == 2
        apply!(clients[i][1], Z)
    end
end
@info "reference state: $(canonicalize!(refstate))"

generated_state = clients[1].staterefs[1].state[]
@info "generated state: $(canonicalize!(generated_state))"
@info "clients indices $([clients[i].stateindices[1] for i in 1:n])"

@info "ORDER STATE"
current_order = [clients[i].stateindices[1] for i in 1:n]
order_state!(generated_state, current_order)
@info "generated state: $(canonicalize!(generated_state))"
@info "Fidelity: $(abs(Ket(canonicalize!(refstate))'*Ket(canonicalize!(generated_state)))^2)"
```

Output
```julia
┌ Info: reference state: + XZ_Z
│ + ZXZ_
│ + _ZXZ
└ + Z_ZX
┌ Info: generated state: 𝒟ℯ𝓈𝓉𝒶𝒷
│ + Z___
│ + _Z__
│ + __Z_
│ + ___Z
│ 𝒮𝓉𝒶𝒷━━
│ + XZZ_
│ + ZX_Z
│ + Z_XZ
└ + _ZZX
[ Info: clients indices [1, 2, 4, 3]
[ Info: ORDER STATE
[ Info: current order [1, 2, 4, 3]
[ Info: swapped indices (3, 4) to get Graphs.SimpleGraphs.SimpleEdge{Int64}[Edge 1 => 2, Edge 1 => 4, Edge 2 => 3, Edge 3 => 4]
[ Info: current order [1, 2, 3, 4]
┌ Info: generated state: 𝒟ℯ𝓈𝓉𝒶𝒷
│ + Z___
│ + _Z__
│ + __Z_
│ + ___Z
│ 𝒮𝓉𝒶𝒷━━
│ + XZ_Z
│ + ZXZ_
│ + _ZXZ
└ + Z_ZX
[ Info: Fidelity: 0.9999999999999984
```

Krastanov, 2025-04-08:
This functionality does not exist right now. One main point is that `refstate` is private and should never be used in user code -- it exists to make simulating with different backends possible and to make it unnecessary for the user to manually track what subsystem they are working on.

Given that the user should never even know about the existence of `refstate` it is better to not make functions that modify, especially not in a way that will then make the rest of the simulation wrong because the indexing of various registers is not matching the state subsystems anymore.

**However, we can make a tool that makes copies of these internal states** for use with external tools. Something like `get_state(regA[1], regB[2], regC[3])` will give us the quantum state of these three slots within three different registers. **Would that fit your usecase?**

Another question is what to do if the state is actually a 4 qubit state that also involves some other register. Do we just trace over subsystems that were not mentioned?

Luisenden, 2025-04-08:
With `refstate` I assume you mean the fields `staterefs` and `stateindices` of a register right?
Gotcha!

Such a tool would definitely be useful (e.g., to glimpse behind the scenes of the register). To keep it simple we could restrict the function to only return the state when the complete list of involved register slots is passed, e.g., the function takes indices: `get_state(regA, regB, regC; inds=[1,2,3])` then using sth like `@assert allequal([inputregs[i].staterefs[j] for (i,j) in enumerate(inds)])` checks if they all point to the same state and  `@assert Set([inputregs[i].stateindices[j] for (i,j) in enumerate(inds)]) == Set(1:length(inputregs))` makes sure the list is complete (i.e., each index is represented by one input register slot).

**For our use case though we would like to keep using the states stored in the register** (as there is time dependent noise associated with it and we interact with it later on when measuring observables etc.) **Is there any other way to compare two states "index" by "index"?** I understand that the above `order_state!` breaks privacy rules, maybe we can use the register slots to order the qubits according to their indices?

```julia
function order_state!(regs::Vector{RegRef}, current_order::Vector{Int})
    # Loop over each index 
    for i in 1:length(current_order)
        # If the qubit at position i isn't i, swap it with wherever qubit i lives
        while current_order[i] != i
            # Find which position holds the qubit i
            correct_index = findfirst(==(i), current_order)

            # Swap the register qubits physically
            apply!((regs[i], regs[correct_index]), sCNOT)
            apply!((regs[correct_index], regs[i]), sCNOT)
            apply!((regs[i], regs[correct_index]), sCNOT)
            current_order[i], current_order[correct_index] = current_order[correct_index], current_order[i]
        end
    end
end
```

Luisenden, 2025-04-08:
Ah but this would still need the private `stateindices` ...

Krastanov, 2025-04-08:
I think there is some more philosophical disagreement about how code is supposed to be structured here. Maybe an XY problem https://en.wikipedia.org/wiki/XY_problem

Could you describe why you want to be able to compare these states? What is the final goal.

Luisenden, 2025-04-08:
Yeah I figured haha. Yes! Let's assume we have two 4-qubit stabilizer states each stored in a register:

- `regA` holds `XZ_Z ZXZ_ _ZXZ Z_ZX` (cycle graph) 
- `regB` holds `XZZ_ ZX_Z Z_XZ _ZZX'

These two stab states seem to be completely different from each other; and when calculating the fidelity between the two the value is always < 1 (using e.g., `observable(regA[1:4], projector(stateB))`) .

However, when we compare their internal state indices

- `regA.stateindices = [1,2,3,4]`
- `regB.stateindices = [1,2,4,3]` 

and realize that these states are actually physically the same, but 'displayed' in the order of the register slots. Using this knowledge we can swap qubits (3,4) in regB and calculate the fidelity which is then 1.

The goal is to check if two states are physically the same. In other words, we want to know if the tensor product of two states is the same, when it is ordered according to the states' indices is the same. Does this make sense?

Krastanov, 2025-04-08:
> The goal is to check if two states are physically the same.

The internal state indices are an important part of the representation. If the observables differ due to some permutation of the state indices, the states do physically differ (e.g. for the experimentalist).

---

I want to focus on `observable(regA[1:4], projector(stateB))` -- where does `stateB` come from? Was it extracted from a register? If that is the case, that would be the underlying issue.

---

> In other words, we want to know if the tensor product of two states is the same, when it is ordered according to the states' indices. Does this make sense?

I did not get that line.

Krastanov, 2025-04-08:
Ah, I figured out why I am confused. There is a bug in QuantumSavory and you are basically trying to work around the bug thinking that it is done so on purpose.

```
julia> initialize!((reg[2],reg[1]), StabilizerState("ZI IX"));

julia> observable(reg[1:2], projector(StabilizerState("XI IZ"))) # should be 1
0.5

julia> observable(reg[1:2], projector(StabilizerState("ZI IX")))
1.0
```

Let me go ahead and try to fix it.

Krastanov, 2025-04-09:
fixed in #212

Luisenden, 2025-04-09:
I merged the master into the piecewise-graphstate branch -- it now works with `QuantumOpticsRepr` but breaks with the `CliffordRepr`  at least with my setup. The function `permutesystems` is part of the `QuantumInterface` and didn't cause any problems before the merge, so maybe I have to update the Manifest as well?

```Julia
julia> using QuantumSavory

julia> reg = Register(2)
Register with 2 slots: [ Qubit | Qubit ]
  Slots:
    nothing
    nothing

julia> initialize!((reg[2],reg[1]), StabilizerState("ZI IX"));

julia> observable(reg[1:2], projector(StabilizerState("XI IZ"))) # should be 1
0.999999999999999 + 0.0im

julia> reg = Register(2, CliffordRepr())
Register with 2 slots: [ Qubit | Qubit ]
  Slots:
    nothing
    nothing

julia> initialize!((reg[2],reg[1]), StabilizerState("ZI IX"));

julia> observable(reg[1:2], projector(StabilizerState("XI IZ"))) # should be 1
ERROR: MethodError: no method matching permutesystems(::QuantumClifford.MixedDestabilizer{QuantumClifford.Tableau{Vector{…}, Matrix{…}}}, ::Vector{Int64})

Closest candidates are:
  permutesystems(::QuantumOpticsBase.LazyProduct, ::Vector{Int64})
   @ QuantumOpticsBase ~/.julia/packages/QuantumOpticsBase/85m2i/src/operators_lazyproduct.jl:71
  permutesystems(::T, ::Any) where T<:QuantumOpticsBase.Ket
   @ QuantumOpticsBase ~/.julia/packages/QuantumOpticsBase/85m2i/src/states.jl:102
  permutesystems(::QuantumOpticsBase.LazyTensor, ::Any)
   @ QuantumOpticsBase ~/.julia/packages/QuantumOpticsBase/85m2i/src/operators_lazytensor.jl:211
  ...

Stacktrace:
 [1] observable(state::QuantumClifford.MixedDestabilizer{…}, indices::Vector{…}, operation::SProjector)
   @ QuantumSavory ~/Documents/github/QuantumSavory.jl/src/backends/clifford/express.jl:26
 [2] observable(regs::Vector{Register}, indices::Vector{Int64}, obs::SProjector; something::Nothing, time::Nothing)
   @ QuantumSavory ~/Documents/github/QuantumSavory.jl/src/baseops/observable.jl:15
 [3] observable
   @ ~/Documents/github/QuantumSavory.jl/src/baseops/observable.jl:8 [inlined]
 [4] observable(refs::Vector{RegRef}, obs::SProjector)
   @ QuantumSavory ~/Documents/github/QuantumSavory.jl/src/baseops/observable.jl:17
 [5] top-level scope
   @ REPL[8]:1
Some type information was truncated. Use `show(err)` to see complete types.
```

Luisenden, 2025-04-09:
I will reopen the issue, as updating did not resolve it unfortunately

closed 2025-04-09 by Luisenden

Krastanov, 2025-04-09:
The thanks should be to you! I imagine detecting and debugging this issue was a major pain, thanks for going through it!