QuantumSavory/QuantumSavory.jl #216: observable() comparatively slow when using QuantumOpticsRepr()

Luisenden, 2025-04-15:
When comparing `observable()` with e.g., NetSquid's `netsquid.qubits.qubitapi.fidelity` one sees quite some performance gap (see attached figure), which is surprising

**Expected behavior**

I would expect Julia to be capable of better performance, as both use sparse matrix representations..

**Minimal Reproducible Example 👇**

```julia
using QuantumSavory
using QuantumClifford: Stabilizer, ghz

times = []
for n in 2:12 # number of qubits
    reg = Register(n)

    ghz_state = StabilizerState(Stabilizer(ghz(n)))

    initialize!(reg[1:n], reduce(⊗, fill(X1, n)))
    for i in 2:n
        apply!((reg[1], reg[i]), ZCZ)
    end
    apply!(reg[1], H)

    timed = @elapsed observable(reg[1:n], projector(ghz_state))
    @info "Time to measure GHZ state with $(n) qubits: $(timed)"
    push!(times, timed)
end

@info "Times: $(times)"
```
and
(using `netsquid==0.10.3`)

```python
from time import time
import numpy as np
import netsquid as ns
import netsquid.qubits.qubitapi as qapi
from netsquid.qubits import operators as ops

ns.set_qstate_formalism(ns.QFormalism.SPARSEDM)
depolar_rate = 0.1

times = []

for nqubits in range(2, 13):

    qubits = qapi.create_qubits(nqubits)
    for i in range(nqubits):
        qapi.operate(qubits[i], ops.H)
    for j in range(1,nqubits):
        qapi.operate([qubits[0], qubits[j]], ops.CZ)
    qapi.operate(qubits[0], ops.H)

    def create_ghz_state(num_qubits):
        dim = 2 ** num_qubits
        ghz = np.zeros(dim)
        ghz[0] = 1 / np.sqrt(2)
        ghz[dim - 1] = 1 / np.sqrt(2)
        return ghz

    ghz = create_ghz_state(nqubits)

    start = time()
    fidel = qapi.fidelity(qubits, ghz, squared=True)
    timed = time() - start
    print(f"Fidelity: {fidel}")
    times.append(timed)
print(times)
```

**Environment (please complete the following information):**

  - Output of `using Pkg; Pkg.status()`

```julia
Status `~/Documents/github/QuantumSavory.jl/examples/graphstateswitch/Project.toml`
  [c7e460c6] ArgParse v1.2.0
  [336ed68f] CSV v0.10.15
  [861a8166] Combinatorics v1.0.2
  [6ed1e86c] ConcurrentSim v1.5.0
  [8f4d0f93] Conda v1.10.2
  [a93c6f00] DataFrames v1.7.0
  [31c24e10] Distributions v0.25.118
  [bd48cda9] GraphRecipes v0.5.13
  [86223c79] Graphs v1.12.1
  [c8e1da08] IterTools v1.10.0
  [b964fa9f] LaTeXStrings v1.4.0
  [46757867] NetworkLayout v0.4.10
  [f0f68f2c] PlotlyJS v0.18.15
⌃ [91a5bcdd] Plots v1.40.11
  [438e738f] PyCall v1.96.4
  [0525e862] QuantumClifford v0.9.19
  [5717a53b] QuantumInterface v0.3.8
  [4f57444f] QuantumOpticsBase v0.5.5
  [2de2e421] QuantumSavory v0.6.0 `../..`
  [efa7fd63] QuantumSymbolics v0.4.8
  [158674fc] ReinforcementLearning v0.11.0
  [c5292f4c] ResumableFunctions v1.0.3
  [295af30f] Revise v3.7.3
  [2913bbd2] StatsBase v0.34.4
  [f3b207a7] StatsPlots v0.15.7
  [ddb6d928] YAML v0.4.13
  [8dfed614] Test v1.11.0
```

  - Output of `versioninfo()`

```julia
Julia Version 1.11.4
Commit 8561cc3d68d (2025-03-10 11:36 UTC)
Build Info:
  Official https://julialang.org/ release
Platform Info:
  OS: macOS (x86_64-apple-darwin24.0.0)
  CPU: 10 × Apple M2 Pro
  WORD_SIZE: 64
  LLVM: libLLVM-16.0.6 (ORCJIT, westmere)
Threads: 1 default, 0 interactive, 1 GC (on 10 virtual cores)
```

[ghz_state_measurement.pdf](https://github.com/user-attachments/files/19763289/ghz_state_measurement.pdf)

Krastanov, 2025-04-15:
This will be at least partially addressed in #217 which will be merged shortly.

The main reason for the slowdown is that we were computing Tr(rho* |ψ⟩⟨ψ|) instead of ⟨ψ|rho|ψ⟩. Now we detect the pattern of an observable being a projector and shortcircuit to the faster implementation.

There are probably a few more differences: the `observable` function is capable of dealing with the case of |ψ⟩ living in only a subspace of the entire space of rho, which does not seem to be handled in the netsquid case. Of course, we would want to be faster independently of this, but we would probably need a bit more detailed benchmarking in such a case.

Do you mind trying out your benchmark and reporting timing information as well after #217 is merged?

Luisenden, 2025-04-16:
Ah nice! Sure I will do, let me know when it's ready

Krastanov, 2025-04-16:
should be merged and ready now

Luisenden, 2025-04-17:
I still get the same results -- I just pulled the master, do I need to do anything else before execution?

Krastanov, 2025-04-17:
Just a minor thing:

`StabilizerState(Stabilizer(ghz(n)))` can just be `StabilizerState(ghz(n))`

Krastanov, 2025-04-17:
In the meantime I am also running what you shared in the initial message. With the newest commit:

```
[ Info: Time to measure GHZ state with 2 qubits: 0.435561443
[ Info: Time to measure GHZ state with 3 qubits: 0.130772053
[ Info: Time to measure GHZ state with 4 qubits: 0.23515233
[ Info: Time to measure GHZ state with 5 qubits: 0.318659931
[ Info: Time to measure GHZ state with 6 qubits: 0.443387516
[ Info: Time to measure GHZ state with 7 qubits: 0.578594588
[ Info: Time to measure GHZ state with 8 qubits: 0.601636207
[ Info: Time to measure GHZ state with 9 qubits: 0.776537365
[ Info: Time to measure GHZ state with 10 qubits: 1.244499223
[ Info: Time to measure GHZ state with 11 qubits: 2.38144003
[ Info: Time to measure GHZ state with 12 qubits: 6.659541501
```

On HEAD~1 I get
```
[ Info: Time to measure GHZ state with 2 qubits: 0.629041415
[ Info: Time to measure GHZ state with 3 qubits: 0.229622634
[ Info: Time to measure GHZ state with 4 qubits: 0.314314255
[ Info: Time to measure GHZ state with 5 qubits: 0.407423253
[ Info: Time to measure GHZ state with 6 qubits: 0.539851576
[ Info: Time to measure GHZ state with 7 qubits: 0.661339867
[ Info: Time to measure GHZ state with 8 qubits: 0.703182016
[ Info: Time to measure GHZ state with 9 qubits: 0.935839879
[ Info: Time to measure GHZ state with 10 qubits: 1.432941
[ Info: Time to measure GHZ state with 11 qubits: 2.55703233
[ Info: Time to measure GHZ state with 12 qubits: 7.128065396
```

I confirm this example is indeed still slow. I will look into why it is not going through the new faster pathway.

Krastanov, 2025-04-17:
Ah, it is a silly issue. We were measuring compilation time in the example above, not actual simulation time:

```
for n in 2:12 # number of qubits
    reg = Register(n)

    ghz_state = StabilizerState(Stabilizer(ghz(n)))

    initialize!(reg[1:n], reduce(⊗, fill(X1, n)))
    for i in 2:n
        apply!((reg[1], reg[i]), ZCZ)
    end
    apply!(reg[1], H)

    timed = @elapsed observable(reg[1:n], projector(ghz_state))
    @info "Time to measure GHZ state with $(n) qubits: $(timed)"
    timed = @elapsed observable(reg[1:n], projector(ghz_state))
    @info "Time to measure GHZ state with $(n) qubits second time: $(timed)"
end
```

outputs

```
[ Info: Time to measure GHZ state with 2 qubits: 0.088156025
[ Info: Time to measure GHZ state with 2 qubits second time: 1.4348e-5
[ Info: Time to measure GHZ state with 3 qubits: 0.139805696
[ Info: Time to measure GHZ state with 3 qubits second time: 1.586e-5
[ Info: Time to measure GHZ state with 4 qubits: 0.226216686
[ Info: Time to measure GHZ state with 4 qubits second time: 1.2353e-5
[ Info: Time to measure GHZ state with 5 qubits: 0.000146959
[ Info: Time to measure GHZ state with 5 qubits second time: 2.155e-6
[ Info: Time to measure GHZ state with 6 qubits: 0.458528144
[ Info: Time to measure GHZ state with 6 qubits second time: 1.589e-5
[ Info: Time to measure GHZ state with 7 qubits: 0.576676485
[ Info: Time to measure GHZ state with 7 qubits second time: 1.7123e-5
[ Info: Time to measure GHZ state with 8 qubits: 0.616951441
[ Info: Time to measure GHZ state with 8 qubits second time: 1.7243e-5
[ Info: Time to measure GHZ state with 9 qubits: 0.812004907
[ Info: Time to measure GHZ state with 9 qubits second time: 1.6291e-5
[ Info: Time to measure GHZ state with 10 qubits: 1.236145173
[ Info: Time to measure GHZ state with 10 qubits second time: 1.5459e-5
[ Info: Time to measure GHZ state with 11 qubits: 2.404179074
[ Info: Time to measure GHZ state with 11 qubits second time: 1.4998e-5
[ Info: Time to measure GHZ state with 12 qubits: 6.809666519
[ Info: Time to measure GHZ state with 12 qubits second time: 2.2873e-5
```

So the measurement is actually very fast (hundreds of nanoseconds)

Krastanov, 2025-04-17:
ok, here is the final solution:

Your slowdown was for two reasons:
- unnecessary doing an outer product in `observable` which was fixed by the PR that originally closed this issue
- converting a stabilizer state to a ket (something that was not done in the netsquid code to which we are comparing)

Here is a quick demo of timings related to this:
```
for n in 2:12 # number of qubits
    reg = Register(n)

    ghz_state = StabilizerState(Stabilizer(ghz(n)))

    initialize!(reg[1:n], reduce(⊗, fill(X1, n)))
    for i in 2:n
        apply!((reg[1], reg[i]), ZCZ)
    end
    apply!(reg[1], H)

    p = projector(ghz_state)
    timed = @elapsed express(ghz_state) # this gets cached so the next time it is much faster, including its use in observable
    @info "Time to express GHZ state with $(n) qubits: $(timed)"
    timed = @elapsed observable(reg[1:n], p)
    @info "Time to measure GHZ state with $(n) qubits: $(timed)"
    timed = @elapsed observable(reg[1:n], p)
    @info "Time to measure GHZ state with $(n) qubits second time: $(timed)"

    alternative_ghz_state = (tensor(fill(L0,n)...)+tensor(fill(L1,n)...))/sqrt(2)
    timed = @elapsed express(alternative_ghz_state) # this gets cached so the next time it is much faster, including its use in observable
    @info "Time to express GHZ state with $(n) qubits alternative: $(timed)"
end
```

```
[ Info: Time to express GHZ state with 2 qubits: 1.9107e-5
[ Info: Time to measure GHZ state with 2 qubits: 8.877e-6
[ Info: Time to measure GHZ state with 2 qubits second time: 1.412e-6
[ Info: Time to express GHZ state with 2 qubits alternative: 2.7433e-5
[ Info: Time to express GHZ state with 3 qubits: 3.9766e-5
[ Info: Time to measure GHZ state with 3 qubits: 4.328e-6
[ Info: Time to measure GHZ state with 3 qubits second time: 8.02e-7
[ Info: Time to express GHZ state with 3 qubits alternative: 4.0988e-5
[ Info: Time to express GHZ state with 4 qubits: 9.5912e-5
[ Info: Time to measure GHZ state with 4 qubits: 6.672e-6
[ Info: Time to measure GHZ state with 4 qubits second time: 1.433e-6
[ Info: Time to express GHZ state with 4 qubits alternative: 2.4847e-5
[ Info: Time to express GHZ state with 5 qubits: 0.000171336
[ Info: Time to measure GHZ state with 5 qubits: 4.809e-6
[ Info: Time to measure GHZ state with 5 qubits second time: 1.072e-6
[ Info: Time to express GHZ state with 5 qubits alternative: 2.0719e-5
[ Info: Time to express GHZ state with 6 qubits: 0.000469502
[ Info: Time to measure GHZ state with 6 qubits: 5.48e-6
[ Info: Time to measure GHZ state with 6 qubits second time: 2.394e-6
[ Info: Time to express GHZ state with 6 qubits alternative: 3.0347e-5
[ Info: Time to express GHZ state with 7 qubits: 0.001646139
[ Info: Time to measure GHZ state with 7 qubits: 5.481e-6
[ Info: Time to measure GHZ state with 7 qubits second time: 1.423e-6
[ Info: Time to express GHZ state with 7 qubits alternative: 0.008791707
[ Info: Time to express GHZ state with 8 qubits: 0.014907971
[ Info: Time to measure GHZ state with 8 qubits: 1.2023e-5
[ Info: Time to measure GHZ state with 8 qubits second time: 1.683e-6
[ Info: Time to express GHZ state with 8 qubits alternative: 4.3082e-5
[ Info: Time to express GHZ state with 9 qubits: 0.050093976
[ Info: Time to measure GHZ state with 9 qubits: 1.2103e-5
[ Info: Time to measure GHZ state with 9 qubits second time: 1.963e-6
[ Info: Time to express GHZ state with 9 qubits alternative: 4.9504e-5
[ Info: Time to express GHZ state with 10 qubits: 0.250657989
[ Info: Time to measure GHZ state with 10 qubits: 1.6462e-5
[ Info: Time to measure GHZ state with 10 qubits second time: 2.635e-6
[ Info: Time to express GHZ state with 10 qubits alternative: 5.8661e-5
[ Info: Time to express GHZ state with 11 qubits: 1.215834416
[ Info: Time to measure GHZ state with 11 qubits: 1.7794e-5
[ Info: Time to measure GHZ state with 11 qubits second time: 3.056e-6
[ Info: Time to express GHZ state with 11 qubits alternative: 7.3049e-5
[ Info: Time to express GHZ state with 12 qubits: 5.598914077
[ Info: Time to measure GHZ state with 12 qubits: 1.8996e-5
[ Info: Time to measure GHZ state with 12 qubits second time: 5.059e-6
[ Info: Time to express GHZ state with 12 qubits alternative: 7.7407e-5
```

As you can see, the only slow part now is converting a stabilizer state to a ket vector. One way to get around this slowdown and still stick to your code is to do `const ghz5 = StabilizerState(ghz(5))` or even `const ghzs = [StabilizerState(ghz(n)) for n in 1:10]` and then just reuse that constant, instead of creating a new ghz state each time you are running an observable. Any conversions will be cached, so as long as you stick to not creating brand new states each time, things should be fast.

Closing this for now, but let me know if anything else comes up. Let me know if this addresses your issue.

closed 2025-04-17 by Krastanov

Krastanov, 2025-04-28:
and thanks for bringing this up! Either way, it led to some improvements to performance in `observable` as discussed in the related PR

PR #217 (2025-04-16): Speedup observable by avoiding outer products
Fixes #216 (at least partially -- it is a speadup of more than 10x for 5-qubit states)

```
BenchmarkTools.Trial: 10000 samples with 1 evaluation per sample.
 Range (min … max):   5.831 μs …   4.401 ms  ┊ GC (min … max):  0.00% … 99.13%
 Time  (median):      8.336 μs               ┊ GC (median):     0.00%
 Time  (mean ± σ):   13.733 μs ± 147.468 μs  ┊ GC (mean ± σ):  36.98% ±  3.43%

      ▁▃▄▄▄▄▇▄▆▆▇▆▆█▅▅▄▄▂▁▁
  ▁▂▅███████████████████████▇▆▆▄▅▄▃▃▃▃▃▃▃▂▂▂▂▂▂▂▂▂▂▂▂▁▂▂▁▁▁▁▁▁ ▄
  5.83 μs         Histogram: frequency by time           15 μs <

 Memory  [... cut]